neuropsychology/NeuroKit #866: ECG output missing using epochs_create

Likleung18, 2023-07-14:
**Question and context**

Hi all, 

I have used event_create and epochs_create functions to extract segments of clean signal from ECG recordings. For each recording I would identify 2-4 segments, the issue I am facing is that there may be nothing shown in the DataFrame for some epochs, as this example shows:
 
[image]

THe sample data can be found [here](https://connecthkuhk-my.sharepoint.com/:u:/g/personal/lik77459_connect_hku_hk/EQSfMYVZlQpNr__RwL6zq2wBxgzTe1bFHLvHVHYuKngpUg?e=sMzy6F) and below is my code for segments extraction and subsequent analysis.
```
#%%
import neurokit2 as nk

#%%
data_R014_Resting_2, sampling_rate = nk.read_acqknowledge('C:/Users/Psychology/Desktop/ECG/R014/R014_resting_20201030.acq',impute_missing=True)
data_R014_Resting_2_signals, info_R014_Resting_2_signals = nk.ecg_process(data_R014_Resting_2['ECG, Y, RSPEC-R'], sampling_rate=2000)
R014_Resting_2_events = nk.events_create(event_onsets = [152000], event_durations=306000,event_labels=['Resting_2'])
R014_Resting_2_epochs=nk.epochs_create(data_R014_Resting_2_signals,events=R014_Resting_2_events,sampling_rate=2000,epochs_start=76,epochs_end=229)
R014_Resting_2_results = nk.bio_analyze(R014_Resting_2_epochs,sampling_rate=2000)

#%%
data_R014_Resting_3, sampling_rate = nk.read_acqknowledge('C:/Users/Psychology/Desktop/ECG/R014/R014_resting_20201030.acq',impute_missing=True)
data_R014_Resting_3_signals, info_R014_Resting_3_signals = nk.ecg_process(data_R014_Resting_3['ECG, Y, RSPEC-R'], sampling_rate=2000)
R014_Resting_3_events = nk.events_create(event_onsets = [504000], event_durations=98000,event_labels=['Resting_3'])
R014_Resting_3_epochs=nk.epochs_create(data_R014_Resting_3_signals,events=R014_Resting_3_events,sampling_rate=2000,epochs_start=252,epochs_end=301)
R014_Resting_3_results = nk.bio_analyze(R014_Resting_3_epochs,sampling_rate=2000)
```
I would be grateful if anyone could guide me on how to resolve this issue. I am also not sure if it is the right way to do segments extraction. If possible, could you please provide me with a better way to do so?

Many thanks!

Lik

Likleung18, 2023-07-19:
The issue is tackled.

DominiqueMakowski, 2023-07-19:
Good to hear.  Would you mind sharing the isssue & solution for other potential users?

Likleung18, 2023-07-19:
Just found that it was my misunderstandings of using epochs_start and epochs_end. It was tackled by resetting their values.