neuropsychology/NeuroKit #869: Error in basSQI calculation (See ecg_quality.py)?

Abdul099, 2023-07-16:
Commit: f6d4d2d by @Jacob-Liam-Albaugh 

The formula used to calculate basSQI was updated recently from 1 - (num_power/dem_power) to (1-num_power)/dem_power, where num_power is the psd of the frequencies 0-1Hz and dem_power is the psd of the frequencies 0-40Hz). 

The aim of the SQI is to evaluate the baseline wander in the ECG signal. According to the referenced paper (Zhao 2018), an optimal basSQI lies between 0.95 and 1 whereas a poor basSQI is below 0.95. 

Using the old formula, if we assume an ideal signal has zero psd for 0-1Hz and some value of psd for 0-40Hz, we would get a basSQI of 1-(0/dem_power) = 1. However, with the new formula, we would get a basSQI value of (1-0)/dem_power = 1/dem_power, which could evaluate to a number close to 0 depending how big dem_power is. 

The formula in the referenced paper is in fact the new one :(1-num_power)/dem_power. However, this does not align with the purpose of the index itself.

Thank you in advance!

DominiqueMakowski, 2023-07-17:
If I understand, you would recommend reverting to the old one?

Abdul099, 2023-07-18:
@DominiqueMakowski exactly. You can also check this paper where basSQI is calculated as P(1-40)/P(0-40): https://www.sciencedirect.com/science/article/pii/S1746809419302472

DominiqueMakowski, 2023-07-26:
Then we should probably do that if that makes sense, but document the rationale as a comment in the code + in the docstrings

closed 2024-03-17 by stale[bot]