neuropsychology/NeuroKit #895: nk.ppg_plot() not working even with example data

Mleminator, 2023-09-04:
I'm trying to use the ppg functions but on the first example, where a ppg signal is generated, processed and then plotted, the nk.ppg_plot() function gives the following error:

File "C:\Users\maxsu\AppData\Local\Programs\Python\Python310\lib\site-packages\neurokit2\ppg\ppg_plot.py", line 60, in ppg_plot
    x_axis = np.linspace(0, ppg_signals.shape[0] / sampling_rate, ppg_signals.shape[0])
TypeError: unsupported operand type(s) for /: 'int' and 'dict'

DominiqueMakowski, 2023-09-04:
Yes, this is because the examples correspond to the dev version which should be released in a couple of days. You can install the dev version by following the steps in the installation section of the docs

Mleminator, 2023-09-04:
Oh ok thank you! Didn't catch that

closed 2023-09-04 by DominiqueMakowski