neuropsychology/NeuroKit #897: FutureWarning in log of failed docs check due to ecg_delineate

danibene, 2023-09-06:
The docs build is currently failing: https://github.com/neuropsychology/NeuroKit/actions/runs/6081656167/job/16538141261?pr=863
And includes the following warning: 
```
D:\a\NeuroKit\NeuroKit\neurokit2\ecg\ecg_delineate.py:1015: FutureWarning: The behavior of obj[i:j] with a float-dtype index is deprecated. In a future version, this will be treated as positional instead of label-based. For label-based slicing, use obj.loc[i:j] instead
  segment = heartbeat[:0]  # Select left hand side
```

I thought this should be addressed by changing `heartbeat[:0]` to `heartbeat.iloc[:0]`, but this led to a failing test:
https://github.com/neuropsychology/NeuroKit/actions/runs/6098830607/job/16549573373 

So it would be good if someone could eventually look into the best way to address this warning

DominiqueMakowski, 2023-09-07:
Shouldn't it be .loc[0:] ? 

I think it's because the index is a float (corresponding to time) and we want to subset the stuff after the R-peak (that happens at 0 ms)

danibene, 2023-09-07:
> Shouldn't it be .loc[0:] ?
> 
> I think it's because the index is a float (corresponding to time) and we want to subset the stuff after the R-peak (that happens at 0 ms)

Ah yes that makes sense thank you!

closed 2023-09-07 by danibene