neuropsychology/NeuroKit #905: urllib.error.URLError: <urlopen error [Errno 11004] getaddrinfo failed>

xuexiaoxiaohu, 2023-09-15:
I am interested in your code, but when i download data , it has error as follows

my python code :
# Load NeuroKit and other useful packages
import neurokit2 as nk
import numpy as np
import pandas as pd
print('nk.version()： ',nk.version())
# Retrieve ECG data from data folder
ecg_signal = nk.data(dataset="ecg_1000hz")

results as follows:
- OS: Windows (WindowsPE 64bit) 
- Python: 3.9.0 
- NeuroKit2: 0.2.3 

- NumPy: 1.23.5 
- Pandas: 1.5.3 
- SciPy: 1.10.0 
- sklearn: 1.2.1 
- matplotlib: 3.7.1
nk.version()：  None
Traceback (most recent call last):
  File "C:\Users\11024\anaconda3\envs\fam3d\lib\urllib\request.py", line 1342, in do_open
    h.request(req.get_method(), req.selector, req.data, headers,
  File "C:\Users\11024\anaconda3\envs\fam3d\lib\http\client.py", line 1255, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "C:\Users\11024\anaconda3\envs\fam3d\lib\http\client.py", line 1301, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "C:\Users\11024\anaconda3\envs\fam3d\lib\http\client.py", line 1250, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "C:\Users\11024\anaconda3\envs\fam3d\lib\http\client.py", line 1010, in _send_output
    self.send(msg)
  File "C:\Users\11024\anaconda3\envs\fam3d\lib\http\client.py", line 950, in send
    self.connect()
  File "C:\Users\11024\anaconda3\envs\fam3d\lib\http\client.py", line 1417, in connect
    super().connect()
  File "C:\Users\11024\anaconda3\envs\fam3d\lib\http\client.py", line 921, in connect
    self.sock = self._create_connection(
  File "C:\Users\11024\anaconda3\envs\fam3d\lib\socket.py", line 822, in create_connection
    for res in getaddrinfo(host, port, 0, SOCK_STREAM):
  File "C:\Users\11024\anaconda3\envs\fam3d\lib\socket.py", line 953, in getaddrinfo
    for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
socket.gaierror: [Errno 11004] getaddrinfo failed

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "D:\pyProject\ele_loc\FAM\r_detect.py", line 8, in <module>
    ecg_signal = nk.data(dataset="ecg_1000hz")
  File "C:\Users\11024\anaconda3\envs\fam3d\lib\site-packages\neurokit2\data\data.py", line 187, in data
    return pd.read_csv(path + "ecg_1000hz.csv")["ECG"].values
  File "C:\Users\11024\anaconda3\envs\fam3d\lib\site-packages\pandas\util\_decorators.py", line 211, in wrapper
    return func(*args, **kwargs)
  File "C:\Users\11024\anaconda3\envs\fam3d\lib\site-packages\pandas\util\_decorators.py", line 331, in wrapper
    return func(*args, **kwargs)
  File "C:\Users\11024\anaconda3\envs\fam3d\lib\site-packages\pandas\io\parsers\readers.py", line 950, in read_csv
    return _read(filepath_or_buffer, kwds)
  File "C:\Users\11024\anaconda3\envs\fam3d\lib\site-packages\pandas\io\parsers\readers.py", line 605, in _read
    parser = TextFileReader(filepath_or_buffer, **kwds)
  File "C:\Users\11024\anaconda3\envs\fam3d\lib\site-packages\pandas\io\parsers\readers.py", line 1442, in __init__
    self._engine = self._make_engine(f, self.engine)
  File "C:\Users\11024\anaconda3\envs\fam3d\lib\site-packages\pandas\io\parsers\readers.py", line 1735, in _make_engine
    self.handles = get_handle(
  File "C:\Users\11024\anaconda3\envs\fam3d\lib\site-packages\pandas\io\common.py", line 713, in get_handle
    ioargs = _get_filepath_or_buffer(
  File "C:\Users\11024\anaconda3\envs\fam3d\lib\site-packages\pandas\io\common.py", line 363, in _get_filepath_or_buffer
    with urlopen(req_info) as req:
  File "C:\Users\11024\anaconda3\envs\fam3d\lib\site-packages\pandas\io\common.py", line 265, in urlopen
    return urllib.request.urlopen(*args, **kwargs)
  File "C:\Users\11024\anaconda3\envs\fam3d\lib\urllib\request.py", line 214, in urlopen
    return opener.open(url, data, timeout)
  File "C:\Users\11024\anaconda3\envs\fam3d\lib\urllib\request.py", line 517, in open
    response = self._open(req, data)
  File "C:\Users\11024\anaconda3\envs\fam3d\lib\urllib\request.py", line 534, in _open
    result = self._call_chain(self.handle_open, protocol, protocol +
  File "C:\Users\11024\anaconda3\envs\fam3d\lib\urllib\request.py", line 494, in _call_chain
    result = func(*args)
  File "C:\Users\11024\anaconda3\envs\fam3d\lib\urllib\request.py", line 1385, in https_open
    return self.do_open(http.client.HTTPSConnection, req,
  File "C:\Users\11024\anaconda3\envs\fam3d\lib\urllib\request.py", line 1345, in do_open
    raise URLError(err)
urllib.error.URLError: <urlopen error [Errno 11004] getaddrinfo failed>

Process finished with exit code 1

DominiqueMakowski, 2023-09-15:
This looks like an internet issue. Make sure you are connected to the internet when downloading the dataset (maybe disable VPN / firewall too?)

DominiqueMakowski, 2023-09-22:
Did you manage to solve this?

closed 2025-02-01 by stale[bot]