neuropsychology/NeuroKit #907: Who works DWT in ecg_delinate() ??

Adrianpg23, 2023-09-17:
Hi! I don’t understand how works ecg_delinate() when I use “dwt” for method… I saw the code and there are the function “_dwt_compute_multiscales” and inside it, are the functions “_apply_H_filter” and “_apply_G_filter” but I don’t get it, I was searching for the function known as “mother wavelet”. Please, can someone can explain me how the function ecg_delinate() works using dwt? P.d.: Sorry if my English is not good.

DominiqueMakowski, 2023-09-17:
*All hail Mother wavelet, help us uncover thyne secrets* 🙌 🙌 

I think you meant morlet wavelet. I'm afraid I can't help you understand the inner workings of this function, as it's a complex one. But you have access to the code so all I can say is good luck!

closed 2023-09-17 by DominiqueMakowski

Adrianpg23, 2023-09-17:
Hello again, what I want is to know how the dwt method that was implemented in this function (ecg_delinate()) works, that is, what is the mother wavelet function and what do "apply_H_filter" and "_apply_G_filter" do in the calculation of the DWT