neuropsychology/NeuroKit #910: Issue computing ApEn with scikit-learn version 1.3.1

danibene, 2023-09-21:
**Describe the bug**
Tried to run `hrv()` after upgrading from scikit-learn 1.3.0 to 1.3.1 and got the following error:

```
out["ApEn"], _ = entropy_approximate(rri, delay=1, dimension=2, tolerance=tolerance)
/neurokit2/complexity/entropy_approximate.py", line 100, in entropy_approximate
    out, _ = _entropy_apen(signal, delay, dimension, info["Tolerance"], **kwargs)
/neurokit2/complexity/optim_complexity_tolerance.py", line 363, in _entropy_apen
    phi, info = _phi(
/neurokit2/complexity/utils.py", line 29, in _phi
    embedded1, count1, kdtree1 = _get_count(
/neurokit2/complexity/utils.py", line 115, in _get_count
    valid_metrics = sklearn.neighbors.KDTree.valid_metrics() + ["range"]
TypeError: 'list' object is not callable
```
Downgrading to scikit-learn 1.3.0 fixed it.

**System Specifications**
I am using neurokit2 0.2.6, pandas 1.5.3, scipy 1.11.2, and numpy 1.24.3

DominiqueMakowski, 2023-09-21:
Yes, I just realized it's what messing up with the CIs, let me give it a fix

danibene, 2023-09-21:
Seems like it's related to this: https://github.com/scikit-learn/scikit-learn/pull/26754

danibene, 2023-09-21:
> Yes, I just realized it's what messing up with the CIs, let me give it a fix

Thank you!

DominiqueMakowski, 2023-09-21:
Weird the tests still fail as if they weren't running with the update I made

DominiqueMakowski, 2023-09-21:
I think I'll merge that branch soon anyway, and then we'll see from dev

closed 2025-02-01 by stale[bot]