neuropsychology/NeuroKit #915: About recall rsp_analyze function

Hannahhanhan555, 2023-09-28:
This is a template for reporting a bug. You can remove it and write from scratch. These sections are a rough guide, but the important thing is to give enough details so that the developers can **reproduce** the bug on their machine and then investigate.

**Describe the bug**
I want to extract some resp features from resp data.
First,  I process my raw resp data, just show 2 of them. (fs = 64Hz)
`sig_0, info = nk.rsp_process(sig_0, 64)
sig_1, info = nk.rsp_process(sig_1, 64)`
Then I try to extract features from processed data.
`f_0 = nk.rsp_analyze(sig_0, 64)`

The results are like this, f_1 have more features because sig_1 has more resp cycles:
`f_0 = nk.rsp_analyze(sig_0, 64) 
>>> f_0
  RSP_Rate_Mean            RRV_RMSSD            RRV_MeanBB  ... RSP_Phase_Duration_Inspiration RSP_Phase_Duration_Expiration RSP_Phase_Duration_Ratio      
0     21.485973  [518.1048322371641]  [2806.0897435897436]  ...                       1.472656                      1.305288                 1.128223      

[1 rows x 28 columns]`

`f_1 = nk.rsp_analyze(sig_1, 64)
>>> f_1
  RSP_Rate_Mean             RRV_RMSSD            RRV_MeanBB  ... RRV_MFDFA_alpha2_Asymmetry RRV_MFDFA_alpha2_Fluctuation RRV_MFDFA_alpha2_Increment        
0     33.034043  [1331.7795392630542]  [1981.7116477272727]  ...      [-0.5063283533806278]     [0.00010017426967995603]      [0.05214062328627242]        

[1 rows x 37 columns]   `

Until now, nothing goes wrong.
but when I extract features from sig_0 again:
`f_0_new = nk.rsp_analyze(sig_0, 64)
>>> f_0_new
  RSP_Rate_Mean            RRV_RMSSD            RRV_MeanBB  ... RRV_MFDFA_alpha2_Asymmetry RRV_MFDFA_alpha2_Fluctuation RRV_MFDFA_alpha2_Increment
0     21.485973  [518.1048322371641]  [2806.0897435897436]  ...      [-0.5063283533806278]     [0.00010017426967995603]      [0.05214062328627242]

[1 rows x 37 columns]`

The result changed, number of feature is the same as f_1, at the same time, the added feature(MFDFA_aplha2_xxxx) are equal to those of f_1. 

I have tried several times, use different signals, all results are the same.

**Screenshots**

[image]

**System Specifications**

- OS: Windows (WindowsPE 64bit) 
- Python: 3.8.16
- NeuroKit2: 0.2.5

DominiqueMakowski, 2023-09-28:
Sorry, I'm not sure I understood your question

Hannahhanhan555, 2023-10-12:
> Sorry, I'm not sure I understood your question

I mean that for the same input resp signal,  sig_0, under 2 Conditions, has different number of feature. 

Condition 1: 
when I call nk.rsp_analyze() firstly, it can extract 28 features.

Condition 2:
When I first call nk.rsp_analyze() using a longer signal sig_1, 37 features can be extracted.
that is OK, because some of features need a longer input signal.
My question is,
after calling nk.rsp_analyze() using sig_1,  calling nk.rsp_analyze() using sig_0(using in Condition 1),
37 features are extracted,  but in Condition1, only 28 features are extracted.
9 extra features are the same to those features extracted from sig_1.
[image]

DominiqueMakowski, 2023-10-12:
> My question is, after calling nk.rsp_analyze() using sig_1, calling nk.rsp_analyze() using sig_0(using in Condition 1), 37 features are extracted, but in Condition1, only 28 features are extracted. 9 extra features are the same to those features extracted from sig_1

I'm really sorry but I still have trouble to understand what your question is

Hannahhanhan555, 2023-10-13:
> I'm really sorry but I still have trouble to understand what your question is

Now, let's try another way.
Could you please try this code in one terminal and let me know your result?

input signal: 
    sig_0:  resp signal,  length is 7104, sampling rate is  64Hz
    sig_1: resp signal, length is 11520, sampling rate is 64Hz

code:
    sig_0_processed, info = nk.rsp_process(sig_0, 64)
    feature_0 = nk.rsp_analyze(sig_0_processed, 64)
    print(feature_0)
# this is my result.     
[image]

    sig_1_processed, info = nk.rsp_process(sig_1, 64)
    feature_1 = nk.rsp_analyze(sig_1_processed, 64)
    print(feature_1)
# this is my result. 
[image]

    feature_0_new = nk.rsp_analyze(sig_0_processed, 64)
    print(feature_0_new)
# this is my result.
[image]

Just compare feature_0 with  features_0_new. 
You will find this results are different. 
If your results are not like this, please let me know, thank you!

DominiqueMakowski, 2023-10-13:
I can't run this because I don't have the same data as you, and depending on the breathing rate it will detect a different number of cycles which can influence what is then computed. 

If your question is whether it is normal to have a different number of features, as you mentioned it is likely to be related to the length of the signal. If you try with same length signals you should obtain the same amount of features.

In any case, I wouldn't worry too much and just use the "common" features if you want to compare the various signals

closed 2025-02-01 by stale[bot]