neuropsychology/NeuroKit #919: ecg_analyze() bug

dommrichmond, 2023-10-18:
Creating an ecg signal and using ecg analyze results in an error in Neurokit2 (Version 0.2.6)
```
ecg = nk.ecg_simulate(duration=60, sampling_rate=100, heart_rate=80)
signals, info = nk.ecg_process(ecg, sampling_rate=100)
epochs = nk.epochs_create(signals,sampling_rate=100, epochs_end=10)
results = nk.ecg_analyze(epochs,sampling_rate=100)
```

> NeuroKitWarning: DFA_alpha2 related indices will not be calculated. The maximum duration of the windows provided for the long-term correlation is smaller than the minimum duration of windows. Refer to the `scale` argument in `nk.fractal_dfa()` for more information.

DominiqueMakowski, 2023-10-18:
This is not an error but just a warning that some indices will not be calculated due to the short recordings

dommrichmond, 2023-10-18:
Oops, my bad I forgot to paste everything. Following the warning, there is this error:

```
warn(
Traceback (most recent call last):
  File "C:\Users\domin\Documents\Dom\Misc\HRV\HRV.py", line 109, in <module>
    results = nk.ecg_analyze(epochs,sampling_rate=100)
  File "C:\Users\domin\Documents\Dom\Misc\HRV\venv\lib\site-packages\neurokit2\ecg\ecg_analyze.py", line 116, in ecg_analyze
    features = ecg_intervalrelated(data, sampling_rate=sampling_rate)
  File "C:\Users\domin\Documents\Dom\Misc\HRV\venv\lib\site-packages\neurokit2\ecg\ecg_intervalrelated.py", line 83, in ecg_intervalrelated
    intervals[index] = _ecg_intervalrelated_hrv(
  File "C:\Users\domin\Documents\Dom\Misc\HRV\venv\lib\site-packages\neurokit2\ecg\ecg_intervalrelated.py", line 128, in _ecg_intervalrelated_hrv
    results = hrv(rpeaks, sampling_rate=sampling_rate)
  File "C:\Users\domin\Documents\Dom\Misc\HRV\venv\lib\site-packages\neurokit2\hrv\hrv.py", line 107, in hrv
    out.append(hrv_nonlinear(peaks, sampling_rate=sampling_rate))
  File "C:\Users\domin\Documents\Dom\Misc\HRV\venv\lib\site-packages\neurokit2\hrv\hrv_nonlinear.py", line 236, in hrv_nonlinear
    out["ApEn"], _ = entropy_approximate(rri, delay=1, dimension=2, tolerance=tolerance)
  File "C:\Users\domin\Documents\Dom\Misc\HRV\venv\lib\site-packages\neurokit2\complexity\entropy_approximate.py", line 100, in entropy_approximate
    out, _ = _entropy_apen(signal, delay, dimension, info["Tolerance"], **kwargs)
  File "C:\Users\domin\Documents\Dom\Misc\HRV\venv\lib\site-packages\neurokit2\complexity\optim_complexity_tolerance.py", line 363, in _entropy_apen
    phi, info = _phi(
  File "C:\Users\domin\Documents\Dom\Misc\HRV\venv\lib\site-packages\neurokit2\complexity\utils.py", line 29, in _phi
    embedded1, count1, kdtree1 = _get_count(
  File "C:\Users\domin\Documents\Dom\Misc\HRV\venv\lib\site-packages\neurokit2\complexity\utils.py", line 115, in _get_count
    valid_metrics = sklearn.neighbors.KDTree.valid_metrics() + ["range"]
TypeError: 'list' object is not callable
```

dommrichmond, 2023-10-30:
Any ideas on the above?

DominiqueMakowski, 2023-10-30:
Yes, please update neurokit to the dev version and it should be fixed!

closed 2025-02-01 by stale[bot]