neuropsychology/NeuroKit #923: nk.hrv throw a TypeError: 'list' object is not callable

zj2050, 2023-11-02:
when call `hrv_indices = nk.hrv(rpeaks, sampling_rate=64.0, show=DEBUG)`
will take a error like: 'list' object is not callable
rpeaks = `array([ 61, 108, 159, 207, 256, 304, 355, 403, 455, 503, 551, 601, 651,       702, 753, 803, 850, 898, 947], dtype=int64)`

python 3.1.13  and neurokit 0.2.6

DominiqueMakowski, 2023-11-02:
This works for me:

```python
rpeaks = np.array([ 61, 108, 159, 207, 256, 304, 355, 403, 455, 503, 551, 601, 651,       702, 753, 803, 850, 898, 947])
hrv_indices = nk.hrv(rpeaks, sampling_rate=64)
```

(though it throws some warnings)

zj2050, 2023-11-02:
> This works for me:
> 
> ```python
> rpeaks = np.array([ 61, 108, 159, 207, 256, 304, 355, 403, 455, 503, 551, 601, 651,       702, 753, 803, 850, 898, 947])
> hrv_indices = nk.hrv(rpeaks, sampling_rate=64)
> ```
> 
> (though it throws some warnings)

Yes, it can works on our another PC, what's different just python version 3.11.2 and 3.11.3

OS: Windows (WindowsPE 64bit)

Python: 3.11.2

NeuroKit2: 0.2.6

NumPy: 1.23.5

Pandas: 2.1.2

SciPy: 1.10.1

sklearn: 1.3.1

matplotlib: 3.7.1

but run same code in another computer like:

OS: Windows (WindowsPE 64bit)

Python: 3.11.3

NeuroKit2: 0.2.6

NumPy: 1.23.5

Pandas: 2.0.0

SciPy: 1.10.1

sklearn: 1.2.2

matplotlib: 3.7.1

rahulvenugopal, 2023-12-05:
I also faced the same issue. But, error does not pop up when I run time domain and frequency domain. When I run nonlinear then the same error pops up. My ECG time series is ~6 mnts or so. Going to try in older versions of neurokit2 and see

closed 2025-02-01 by stale[bot]