neuropsychology/NeuroKit #927: Missing runtime dependency "requests" in 0.2.7

hoechenberger, 2023-11-06:
Hello! I'm co-maintaining the neurokit2 package on `coda-forge`. When trying to test version 0.2.7 (a simple `import` test), it appeared that a new runtime dependency on `requests` has been introduced, but it's not explicitly specified in the project dependencies. The `requests` dependency often goes unnoticed as most Python installations deploy `requests` by default. Could you please add this package to the list of dependences? Thanks!

DominiqueMakowski, 2023-11-06:
Hi @hoechenberger thanks for mentioning that. Is it enough if I add `requests` to the [test requirements](https://github.com/neuropsychology/NeuroKit/blob/3d004b4777c3e59b288f46b9f0f812ac1e04b5d4/setup.py#L34)?

hoechenberger, 2023-11-06:
Hello @DominiqueMakowski, thanks for the swift response!
No, even `import neurokit2` fails without `requests` installed, so it should most probably go into the list of `requirements`.

Thank you!

aidin36, 2023-12-19:
I ran into this issue too. I hope you release a fix soon!
Thanks for the time you're putting into this package!

The steps to reproduce:

```
mkdir testproject
cd testproject
pipenv install neurokit2
pipenv run python -c "import neurokit2"
```

```
ModuleNotFoundError: No module named 'requests'
```

closed 2025-02-01 by stale[bot]