neuropsychology/NeuroKit #932: epochs_start param in the function epochs_create() is not able to take a list of values

AbhayKoushik, 2023-11-22:
My events is a dict of n events: **{'onset':[.......n], 'duration':[.......n], 'condition':[.......n], 'label': [.......n]}**

When trying to supply a list of length n for `epochs_start` parameter to the function `epochs_create(...)`, 

I get the following error:

> File ~/opt/miniconda3/envs/heartbrain/lib/python3.9/site-packages/neurokit2/epochs/epochs_create.py:157, in epochs_create(data, events, sampling_rate, epochs_start, epochs_end, event_labels, event_conditions, baseline_correction)
>     155 # Find the maximum numbers of samples in an epoch
>     156 parameters["duration"] = list(np.array(parameters["end"]) - np.array(parameters["start"]))
> --> 157 epoch_max_duration = int(max((i * sampling_rate for i in parameters["duration"])))
>     159 # Extend data by the max samples in epochs * NaN (to prevent non-complete data)
>     160 length_buffer = epoch_max_duration
> 
> ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()

It however works with a single number input for epochs_start, for example: epochs_start = 0.7 works but epochs_start = [-0.3,0.2,0.1, 0.4, ..... n values] gives the above error. What should I do?

epochs_end also works with a single number input but not with a list as specified in [this documentation page](https://neuropsychology.github.io/NeuroKit/functions/epochs.html)

**System Specifications**
- OS: Darwin ( 64bit) 
- Python: 3.9.12 
- NeuroKit2: 0.2.7 
- NumPy: 1.24.3 
- Pandas: 1.5.3 
- SciPy: 1.9.1 
- sklearn: 1.1.1 
- matplotlib: 3.4.3

I would be really helpful to resolve this as it's an important function parameter, thanks a lot in advance!

helenkkkk, 2023-12-08:
Hi! Did you fix the problem? I think I stuck in the exactly same place

DominiqueMakowski, 2023-12-08:
Mmh, this works on my side:

```python
import neurokit2 as nk

# Get data
data = nk.data("bio_eventrelated_100hz")

# Find events
events = nk.events_find(data["Photosensor"],
                        threshold_keep='below',
                        event_conditions=["Negative", "Neutral", "Neutral", "Negative"])

epochs = nk.epochs_create(data, events, sampling_rate=100,
                          epochs_start=[0, -1, -1, 0],
                          epochs_end=1)

helenkkkk, 2023-12-09:
Thanks a lot, it's fixed!!! Thanks again

closed 2025-02-01 by stale[bot]