neuropsychology/NeuroKit #934: Getting respiration amplitudes separately (one by one)

melisasaygin, 2023-11-27:
**Question and context**
I do have my rsp processed dataframe with RSP_Clean, RSP_Peaks etc. I added my task labels as a new column to it and I calculate per task respiratory features (images below). The thing is I also need to calculate standard deviations of the respiratory amplitudes as well as the inspiratory durations, expiratory durations, insp-exp duration ratios. So I need the per breath amplitudes and time durations. How can I obtain that? Is there an easy/straightforward way to do so? I want my different features to be consistent so I would prefer somehow adjusting something with the function definitions/returns and getting these directly from neurokit. I tried the rsp_amplitude function but it gives a value per row which is not what I want. For instance, if there are three breath cycles within a task, I want to get each ones amplitude, insp time, exp time, separately. I also need the expiratory amplitude per breath (I saw that one of the methods to get out the volume is an average between the insp and exp volumes so hopefully that can be adjusted to get out those values).

[image]

[image]
[image]

DominiqueMakowski, 2023-11-28:
You can filter the dataframe to keep the values of RSP_Amplitude etc. at peaks (for instance when RSP_Peak == 1). this will give you the amplitude at the peak or trough of each cycle

melisasaygin, 2023-12-07:
I did that both for the inspiratory and expiratory amplitudes separately. Let's say there are 4 full breath cycles within a segment. I then average the amplitudes at peaks to get the mean amplitude over time, but it is slightly different than what Mean_RSP_Amplitude from intervalrelated function gives me. I could not find exactly where Mean_RSP_Amplitude feature is being calculated (the code for this). I was wondering if it, instead of taking the average of amplitudes at the peaks (given that the previous trough of the first peak is also within the segment) takes average of the amplitude column values over the segment (which would not be just a mean of the inspiratory volumes but would be weighted by the duration of breathing cycles). Which way is being used to come up with this feature? Also, I saw in some script of neurokit2 that the Phase is being used rather than the peak ==1 and trough ==1 identifiers. Does that make a difference too?

closed 2025-02-01 by stale[bot]