neuropsychology/NeuroKit #957: G

youngjaeheo2002, 2024-01-27:
**Question and context**

youngjaeheo2002, 2024-01-27:
I also tried to run this 

```
data = nk.ppg_simulate(duration = 10,sampling_rate = 64)

        df,info = nk.ppg_process(data,sampling_rate =64)
        results = nk.ppg_analyze(data = df, sampling_rate = 64)
```

and I got this error in my notebook

```
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Cell In[20], [line 50](vscode-notebook-cell:?execution_count=20&line=50)
     [46](vscode-notebook-cell:?execution_count=20&line=46) data = nk.ppg_simulate(duration = 10,sampling_rate = 64)
     [49](vscode-notebook-cell:?execution_count=20&line=49) df,info = nk.ppg_process(data,sampling_rate =64)
---> [50](vscode-notebook-cell:?execution_count=20&line=50) results = nk.ppg_analyze(data = df, sampling_rate = 64)
     [51](vscode-notebook-cell:?execution_count=20&line=51) '''
     [52](vscode-notebook-cell:?execution_count=20&line=52) df, info = nk.ppg_process(data,sampling_rate = 100)
     [53](vscode-notebook-cell:?execution_count=20&line=53) nk.ppg_analyze(data = df,sampling_rate = 100)
     [54](vscode-notebook-cell:?execution_count=20&line=54) '''
     [55](vscode-notebook-cell:?execution_count=20&line=55) break   

File [~/ENV/lib/python3.10/site-packages/neurokit2/ppg/ppg_analyze.py:113](https://vscode-remote+ssh-002dremote-002bcedar-002ealliancecan-002eca.vscode-resource.vscode-cdn.net/home/heoyoun1/projects/def-alim/heoyoun1/ApneaProject/muse_feature_extraction/~/ENV/lib/python3.10/site-packages/neurokit2/ppg/ppg_analyze.py:113), in ppg_analyze(data, sampling_rate, method)
    [111](https://vscode-remote+ssh-002dremote-002bcedar-002ealliancecan-002eca.vscode-resource.vscode-cdn.net/home/heoyoun1/projects/def-alim/heoyoun1/ApneaProject/muse_feature_extraction/~/ENV/lib/python3.10/site-packages/neurokit2/ppg/ppg_analyze.py:111)     duration = len(data) / sampling_rate
    [112](https://vscode-remote+ssh-002dremote-002bcedar-002ealliancecan-002eca.vscode-resource.vscode-cdn.net/home/heoyoun1/projects/def-alim/heoyoun1/ApneaProject/muse_feature_extraction/~/ENV/lib/python3.10/site-packages/neurokit2/ppg/ppg_analyze.py:112) if duration >= 10:
--> [113](https://vscode-remote+ssh-002dremote-002bcedar-002ealliancecan-002eca.vscode-resource.vscode-cdn.net/home/heoyoun1/projects/def-alim/heoyoun1/ApneaProject/muse_feature_extraction/~/ENV/lib/python3.10/site-packages/neurokit2/ppg/ppg_analyze.py:113)     features = ppg_intervalrelated(data, sampling_rate=sampling_rate)
    [114](https://vscode-remote+ssh-002dremote-002bcedar-002ealliancecan-002eca.vscode-resource.vscode-cdn.net/home/heoyoun1/projects/def-alim/heoyoun1/ApneaProject/muse_feature_extraction/~/ENV/lib/python3.10/site-packages/neurokit2/ppg/ppg_analyze.py:114) else:
    [115](https://vscode-remote+ssh-002dremote-002bcedar-002ealliancecan-002eca.vscode-resource.vscode-cdn.net/home/heoyoun1/projects/def-alim/heoyoun1/ApneaProject/muse_feature_extraction/~/ENV/lib/python3.10/site-packages/neurokit2/ppg/ppg_analyze.py:115)     features = ppg_eventrelated(data)

File [~/ENV/lib/python3.10/site-packages/neurokit2/ppg/ppg_intervalrelated.py:64](https://vscode-remote+ssh-002dremote-002bcedar-002ealliancecan-002eca.vscode-resource.vscode-cdn.net/home/heoyoun1/projects/def-alim/heoyoun1/ApneaProject/muse_feature_extraction/~/ENV/lib/python3.10/site-packages/neurokit2/ppg/ppg_intervalrelated.py:64), in ppg_intervalrelated(data, sampling_rate)
     [62](https://vscode-remote+ssh-002dremote-002bcedar-002ealliancecan-002eca.vscode-resource.vscode-cdn.net/home/heoyoun1/projects/def-alim/heoyoun1/ApneaProject/muse_feature_extraction/~/ENV/lib/python3.10/site-packages/neurokit2/ppg/ppg_intervalrelated.py:62) if len(rate_cols) == 1:
     [63](https://vscode-remote+ssh-002dremote-002bcedar-002ealliancecan-002eca.vscode-resource.vscode-cdn.net/home/heoyoun1/projects/def-alim/heoyoun1/ApneaProject/muse_feature_extraction/~/ENV/lib/python3.10/site-packages/neurokit2/ppg/ppg_intervalrelated.py:63)     intervals.update(_ppg_intervalrelated_formatinput(data))
---> [64](https://vscode-remote+ssh-002dremote-002bcedar-002ealliancecan-002eca.vscode-resource.vscode-cdn.net/home/heoyoun1/projects/def-alim/heoyoun1/ApneaProject/muse_feature_extraction/~/ENV/lib/python3.10/site-packages/neurokit2/ppg/ppg_intervalrelated.py:64)     intervals.update(*_ppg_intervalrelated_hrv(data, sampling_rate))
     [65](https://vscode-remote+ssh-002dremote-002bcedar-002ealliancecan-002eca.vscode-resource.vscode-cdn.net/home/heoyoun1/projects/def-alim/heoyoun1/ApneaProject/muse_feature_extraction/~/ENV/lib/python3.10/site-packages/neurokit2/ppg/ppg_intervalrelated.py:65) else:
     [66](https://vscode-remote+ssh-002dremote-002bcedar-002ealliancecan-002eca.vscode-resource.vscode-cdn.net/home/heoyoun1/projects/def-alim/heoyoun1/ApneaProject/muse_feature_extraction/~/ENV/lib/python3.10/site-packages/neurokit2/ppg/ppg_intervalrelated.py:66)     raise ValueError(
...
    [333](https://vscode-remote+ssh-002dremote-002bcedar-002ealliancecan-002eca.vscode-resource.vscode-cdn.net/home/heoyoun1/projects/def-alim/heoyoun1/ApneaProject/muse_feature_extraction/~/ENV/lib/python3.10/site-packages/neurokit2/complexity/fractal_dfa.py:333)         "NeuroKit error: the window cannot contain more data points than the time series. Decrease 'scale'."
[... cut ...] muse_feature_extraction/~/ENV/lib/python3.10/site-packages/neurokit2/complexity/fractal_dfa.py:334)     )
    [336](https://vscode-remote+ssh-002dremote-002bcedar-002ealliancecan-002eca.vscode-resource.vscode-cdn.net/home/heoyoun1/projects/def-alim/heoyoun1/ApneaProject/muse_feature_extraction/~/ENV/lib/python3.10/site-packages/neurokit2/complexity/fractal_dfa.py:336) return scale

ValueError: NeuroKit error: the window cannot contain more data points than the time series. Decrease 'scale'.

```

saurabh-kataria, 2024-10-23:
I have similar issues. Cannot analyze my ppg signals unfortunately

saurabh-kataria, 2024-10-24:
My signal is 8 second long which gives that epoch error.
So I now double my signal and get SVD error
`processed, info = nk.ppg_process(np.repeat(x,2), sampling_rate=40)`
`features_dict = nk.ppg_analyze(processed, sampling_rate=40, method='interval-related')`

> /scratch/skatar6/anaconda3/envs/tmp4/lib/python3.9/site-packages/neurokit2/hrv/hrv_nonlinear.py:529: NeuroKitWarning: DFA_alpha2 related indices will not be calculated. The maximum duration of the windows provided for the long-term correlation is smaller than the minimum duration of windows. Refer to the `scale` argu
> ment in `nk.fractal_dfa()` for more information.
>   warn(
> /scratch/skatar6/anaconda3/envs/tmp4/lib/python3.9/site-packages/neurokit2/complexity/entropy_multiscale.py:349: RuntimeWarning: invalid value encountered in scalar divide
>   mse = np.trapz(mse) / len(mse)
> /scratch/skatar6/anaconda3/envs/tmp4/lib/python3.9/site-packages/neurokit2/complexity/optim_complexity_k.py:134: RuntimeWarning: divide by zero encountered in divide
>   normalization = (n - 1) / (np.floor((n - k_subrange) / k).astype(int) * k)
> /scratch/skatar6/anaconda3/envs/tmp4/lib/python3.9/site-packages/neurokit2/complexity/optim_complexity_k.py:135: RuntimeWarning: invalid value encountered in multiply
>   sets = (np.nansum(np.abs(np.diff(sig_values)), axis=1) * normalization) / k
> 
> Intel MKL ERROR: Parameter 6 was incorrect on entry to DGELSD.
> ---------------------------------------------------------------------------
> LinAlgError                               Traceback (most recent call last)
> Cell In[21], line 1
> ----> 1 features_dict = nk.ppg_analyze(processed, sampling_rate=40, method='interval-related')
> 
> File /scratch/skatar6/anaconda3/envs/tmp4/lib/python3.9/site-packages/neurokit2/ppg/ppg_analyze.py:93, in ppg_analyze(data, sampling_rate, method)
>      91 # Interval-related analysis
>      92 elif method in ["interval-related", "interval", "resting-state"]:
> ---> 93     features = ppg_intervalrelated(data, sampling_rate=sampling_rate)
>      95 # Auto
>      96 elif method in ["auto"]:
> 
> File /scratch/skatar6/anaconda3/envs/tmp4/lib/python3.9/site-packages/neurokit2/ppg/ppg_intervalrelated.py:65, in ppg_intervalrelated(data, sampling_rate)
>      63 if len(rate_cols) == 1:
>      64     intervals.update(_ppg_intervalrelated_formatinput(data))
> ---> 65     intervals.update(*_ppg_intervalrelated_hrv(data, sampling_rate))
>      66 else:
>      67     raise ValueError(
>      68         "NeuroKit error: ppg_intervalrelated(): Wrong input,"
>      69         "we couldn't extract heart rate. Please make sure"
>      70         "your DataFrame contains a `PPG_Rate` column."
>      71     )
> 
> File /scratch/skatar6/anaconda3/envs/tmp4/lib/python3.9/site-packages/neurokit2/ppg/ppg_intervalrelated.py:135, in _ppg_intervalrelated_hrv(data, sampling_rate)
>     132 peaks = np.where(data["PPG_Peaks"].values)[0]
>     133 peaks = {"PPG_Peaks": peaks}
> --> 135 results = hrv(peaks, sampling_rate=sampling_rate)
>     137 return results.astype("float").to_dict("records")
> 
> File /scratch/skatar6/anaconda3/envs/tmp4/lib/python3.9/site-packages/neurokit2/hrv/hrv.py:107, in hrv(peaks, sampling_rate, show, **kwargs)
>     105 out.append(hrv_time(peaks, sampling_rate=sampling_rate))
>     106 out.append(hrv_frequency(peaks, sampling_rate=sampling_rate))
> --> 107 out.append(hrv_nonlinear(peaks, sampling_rate=sampling_rate))
>     109 # Compute RSA if rsp data is available
>     110 if isinstance(peaks, pd.DataFrame):
>     File /scratch/skatar6/anaconda3/envs/tmp4/lib/python3.9/site-packages/neurokit2/hrv/hrv_nonlinear.py:300, in hrv_nonlinear(peaks, sampling_rate, show, **kwargs)
>     297 out["RCMSEn"], _ = entropy_multiscale(rri, dimension=2, tolerance=tolerance, method="RCMSEn")
>     299 out["CD"], _ = fractal_correlation(rri, delay=1, dimension=2, **kwargs)
> --> 300 out["HFD"], _ = fractal_higuchi(rri, k_max=10, **kwargs)
>     301 out["KFD"], _ = fractal_katz(rri)
>     302 out["LZC"], _ = complexity_lempelziv(rri, **kwargs)
> 
> File /scratch/skatar6/anaconda3/envs/tmp4/lib/python3.9/site-packages/neurokit2/complexity/fractal_higuchi.py:93, in fractal_higuchi(signal, k_max, show, **kwargs)
>      90     k_values = np.arange(1, k_max + 1)
>      91 else:
>      92     # Compute Higuchi
> ---> 93     slope, intercept, info = _complexity_k_slope(k_max, signal)
>      94     k_values = info["k_values"]
>      95     average_values = info["average_values"]
> 
> File /scratch/skatar6/anaconda3/envs/tmp4/lib/python3.9/site-packages/neurokit2/complexity/optim_complexity_k.py:154, in _complexity_k_slope(kmax, signal, k_number)
>     151 average_values = vectorized_Lk(k_values, signal)
>     153 # Slope of best-fit line through points (slope equal to FD)
> --> 154 slope, intercept = -np.polyfit(np.log(k_values), np.log(average_values), 1)
>     155 return slope, intercept, {"k_values": k_values, "average_values": average_values}
> 
> File /scratch/skatar6/anaconda3/envs/tmp4/lib/python3.9/site-packages/numpy/lib/polynomial.py:669, in polyfit(x, y, deg, rcond, full, w, cov)
>     667 scale = NX.sqrt((lhs*lhs).sum(axis=0))
>     668 lhs /= scale
> --> 669 c, resids, rank, s = lstsq(lhs, rhs, rcond)
>     670 c = (c.T/scale).T  # broadcast scale coefficients
>     672 # warn on rank reduction, which indicates an ill conditioned matrix
> 
[... cut ...]
>    2325     b = zeros(b.shape[:-2] + (m, n_rhs + 1), dtype=b.dtype)
> -> 2326 x, resids, rank, s = gufunc(a, b, rcond, signature=signature, extobj=extobj)
>    2327 if m == 0:
>    2328     x[...] = 0
> 
> File /scratch/skatar6/anaconda3/envs/tmp4/lib/python3.9/site-packages/numpy/linalg/linalg.py:124, in _raise_linalgerror_lstsq(err, flag)
>     123 def _raise_linalgerror_lstsq(err, flag):
> --> 124     raise LinAlgError("SVD did not converge in Linear Least Squares")
> 
> LinAlgError: SVD did not converge in Linear Least Squares

closed 2025-02-25 by stale[bot]