neuropsychology/NeuroKit #962: _find_artifacts and zero threshold from _compute_threshold

pvasilek, 2024-02-07:
Hello,

```
neurokit2\signal\signal_fixpeaks.py:307: RuntimeWarning: divide by zero encountered in divide
  mrrs /= th2
neurokit2\signal\signal_fixpeaks.py:307: RuntimeWarning: invalid value encountered in divide
  mrrs /= th2
```

happens when the signal is very noisy so only a few peaks are detected over a long period.
It is just a warning but may require special handling.

```
- OS: Windows (WindowsPE 64bit)
- Python: 3.12.2
- NeuroKit2: 0.2.7

- NumPy: 1.26.4
- Pandas: 2.2.0
- SciPy: 1.12.0
- sklearn: 1.4.0
- matplotlib: 3.8.2
```

mrpaulstone, 2024-05-06:
I'm also having this issue - I'm wondering if a more descriptive warning saying "noisy signal detected" might be an improvement?

DominiqueMakowski, 2024-05-09:
> I'm wondering if a more descriptive warning saying "noisy signal detected" might be an improvement?

That'd be nice but I'm not sure how to detect that reliably

closed 2025-02-24 by stale[bot]