neuropsychology/NeuroKit #984: ReleasePR workflow failing for v0.2.8

EvanCesanek, 2024-04-16:
Perhaps you are already aware of this, but the latest release v0.2.8 is not deploying to PyPI, so it cannot be installed using `pip install`.

See Actions -> Workflows -> ReleasePR -> Merge pull request #926 from neuropsychology/dev -> [Build and publish Python 🐍 distributions 📦 to PyPI and TestPyPI](https://github.com/neuropsychology/NeuroKit/actions/runs/8579764983/job/23655116491#logs)

Apologies for not resolving this myself, but I am not really set up to test this workflow. However I believe it is a simple fix. The workflow YAML file at https://github.com/neuropsychology/NeuroKit/blob/e3f4469ac48be59a946ee31fececaa8e19a45be5/.github/workflows/releasePR.yml must be updated because the `master` branch of gh-action-pypi-publish has been sunset.

Specifically, the current lines 70-80:

```
            - name: Publish distribution 📦 to Test PyPI
              uses: pypa/gh-action-pypi-publish@master
              with:
                  password: ${{ secrets.TEST_PYPI_PASSWORD }}
                  repository_url: https://test.pypi.org/legacy/

            - name: Publish distribution 📦 to PyPI
              #if: startsWith(github.event.ref, 'refs/tags')
              uses: pypa/gh-action-pypi-publish@master
              with:
                  password: ${{ secrets.PYPI_PASSWORD }}
```

must be modified to:

```
            - name: Publish distribution 📦 to Test PyPI
              uses: pypa/gh-action-pypi-publish@release/v1
              with:
                  password: ${{ secrets.TEST_PYPI_PASSWORD }}
                  repository_url: https://test.pypi.org/legacy/

            - name: Publish distribution 📦 to PyPI
              #if: startsWith(github.event.ref, 'refs/tags')
              uses: pypa/gh-action-pypi-publish@release/v1
              with:
                  password: ${{ secrets.PYPI_PASSWORD }}
```

or some other supported reference, as indicated by the error message in the workflow run logs.

Thank you!

DerAndereJohannes, 2024-04-25:
It would be great to get this fixed, i'm guessing that only a maintainer can realistically make and test these changes, right?

EvanCesanek, 2024-04-25:
Thanks @DerAndereJohannes! I agree. I also looked back at this quickly and realized my thoughts above about the source of this failure were incorrect. Examining the ReleasePR workflow logs from the two Attempts reveals the actual issue:

Attempt #1 successfully deployed to Test PyPI: https://test.pypi.org/project/neurokit2/0.2.8/
However, the next step failed, when deploying to regular PyPI with the following error:
```
ERROR    HTTPError: 400 Bad Request from https://upload.pypi.org/legacy/        
         User 'DominiqueMakowski' does not have a verified primary email        
         address. Please add a verified primary email before attempting to      
         upload to PyPI. See https://pypi.org/help/#verified-email for more     
         information.
```

Unfortunately, even if the verified primary email problem is fixed, we will get a new error on retrying the ReleasePR workflow (see Attempt #2). This is due to the fact that 0.2.8 was successfully deployed to TestPyPI in Attempt #1.
```
ERROR    HTTPError: 400 Bad Request from https://test.pypi.org/legacy/          
         File already exists. See https://test.pypi.org/help/#file-name-reuse   
         for more information.
```

So it looks like two things must be done: (1) the verified primary email problem must be corrected and then (2) a modified version of the ReleasePR workflow must be run that skips the "Publish distribution to TestPyPI" step.

DominiqueMakowski, 2024-05-21:
Should be fixed now

closed 2024-05-21 by DominiqueMakowski