neuropsychology/NeuroKit #988: Test for correlation.py

ujdcodr, 2024-05-08:
**Describe the solution you'd like**
Noticed the low code coverage in the cor() function. A basic test can be added to check the correctness of the correlation functions using known examples.
[Pearson](https://en.wikipedia.org/wiki/Pearson_correlation_coefficient#Geometric_interpretation)
[Spearman](https://en.wikipedia.org/wiki/Spearman%27s_rank_correlation_coefficient#Example)
[Kendall (page 3)](https://personal.utdallas.edu/~herve/Abdi-KendallCorrelation2007-pretty.pdf)

**How could we do it?**
```
# pearson
wiki_example_x = np.array([1, 2, 3, 5, 8])
wiki_example_y = np.array([0.11, 0.12, 0.13, 0.15, 0.18])
assert nx.cor(wiki_example_x,wiki_example_y) == 1

# spearman
wiki_example_x = np.array([106,100,86,101,99,103,97,113,112,110])
wiki_example_y = np.array([7,27,2,50,28,29,20,12,6,17])
rez = nx.cor(wiki_example_x,wiki_example_y,"spearman")
assert np.allclose(rez, -0.175757575, atol=0.0001)

# kendalltau
utd_example_x = np.array([1, 3, 2, 4])
utd_example_y = np.array([1, 4, 2, 3])
rez = nx.cor(utd_example_x,utd_example_y,"kendall")
assert np.allclose(rez, 0.6666666666666669, atol=0.0001)

# tests incorrect inputs for method argument
try:
    rez = nx.cor(wiki_example_x,wiki_example_y,"pearso")
except ValueError as e:
    pass
```

DominiqueMakowski, 2024-05-09:
Sounds good, could you make a PR to add that?

ujdcodr, 2024-05-09:
Sure Dominique. I made the fix in my fork, but I'm unable to run the coverage test locally. I tried the following:

`python3 tests.py tests_stats.py`  

`ERROR: usage: tests.py [options] [file_or_dir] [file_or_dir] [...]  
tests.py: error: unrecognized arguments: --cov --cov-append --cov-report=xml tests_stats.py   
inifile: /home/ujjwal97/nkgit/NeuroKit/pytest.ini  
rootdir: /home/ujjwal97/nkgit/NeuroKit`

I hit the same error with
`python3 tests.py --cov tests_stats.py`

I also didn't find any procedure in the documentation to run coverage tests.

DominiqueMakowski, 2024-05-11:
It's alright just open a PR and the tests should run automatically

ujdcodr, 2024-05-13:
Opened : https://github.com/neuropsychology/NeuroKit/pull/989
Looks like one of the checks is failing and it's not related to my changes.