neuropsychology/NeuroKit #997: NumPy DeprecationWarning thrown in `neurokit2.signal.signal_filter._signal_filter_butterworth`

MarcBalle, 2024-06-11:
**Describe the bug**
When running `neurokit2.ecg_clean` there is the following numpy deprecation warning:
`DeprecationWarning: Conversion of an array with ndim > 0 to a scalar is deprecated, and will error in future. Ensure you extract a single element from your array before performing this operation. (Deprecated NumPy 1.25.)`

Debugging into it, I suspect that the error happens when calling `scipy.signal.butter` in `neurokit2.signal.signal_filter._signal_filter_butterworth`. The `Wn` argument in `scipy.signal.butter`  should be a scalar when `btype` is `lowpass` or `highpass`, but in this case `freqs` is a list.

**To Reproduce**

1. Install neurokit2==0.2.9, scipy==1.13.1, numpy==1.26.4
2. Run `neurokit2.ecg_clean(ecg_signal=np.random.rand(20), sampling_rate=300.0, method="neurokit")` (I use a random sequence here, but it is enough to reproduce the error. It also happens with an actual ecg signal)

**System Specifications**

Running on Windows 10, python 3.9.18. Example:

```python
import neurokit2 as nk
nk.ecg_clean(np.random.rand(20), sampling_rate=300.0, method="neurokit")
```
Warning: 
`DeprecationWarning: Conversion of an array with ndim > 0 to a scalar is deprecated, and will error in future. Ensure you extract a single element from your array before performing this operation. (Deprecated NumPy 1.25.)`

DominiqueMakowski, 2024-06-11:
Hi @MarcBalle thanks! Would you mind making a PR to address this?

MarcBalle, 2024-06-12:
@DominiqueMakowski Done :)

closed 2025-02-24 by stale[bot]