neuropsychology/NeuroKit #1009: ts2vg doesn't support numpy 2.0

ajb5d, 2024-07-12:
During my work on #1007 I noted that CI fails when you call `ecg_findpeaks` with the the vgraph method -- the package `ts2vg` currently requires numpy <2.0 but we don't pin a version and some package solves will pull an incompatible version. ts2vg is only used in `_ecg_findpeaks_visibilitygraph` and isn't included as a general dependency.

ajb5d, 2024-07-12:
A follow up detail -- building ts2vg from source works fine. I think the issue is that the binary wheel that's available needs to be rebuilt against numpy 2.0. I think the ts2vg package needs to be re-published to pypi

DominiqueMakowski, 2024-07-12:
is there anything we can do on NK side?

ajb5d, 2024-07-12:
I think options would be:
1. Pin to numpy < 2.0
1. Adjust CI to install ts2vg from source 

Neither one of these feels super attractive. I opened an issue with ts2vg (https://github.com/CarlosBergillos/ts2vg/issues/36) to see if they can just push a new version.

ajb5d, 2024-07-14:
ts2vg 1.2.4 was released today and fixes this.

closed 2024-07-14 by ajb5d