neuropsychology/NeuroKit #1038: Inconsistency RRV_DFA_alpha1/alpha2 definitions and implementation?

michielmeys, 2024-10-04:
The implementation of when RRV_DFA_alpha1 and RRV_DFA_alpha2 are computed (within neurokit2/rsp/rsp_rrv.py; lines 224 to 241) appears inconsistent with their definitions within the documentation/this function's docstring. Currently, the long-term fluctuation value is computed before even the threshold of the short-term value is reached, for more than 65 cycles, rather than 640.

Could this be verified?

DominiqueMakowski, 2024-10-04:
Could you clarify what you mean? And could you help letting us know what you would change (in the docs or the code)

michielmeys, 2024-10-04:
Sorry for the lack of clarity. It just appears that DFA_alpha2 can currently be computed for breathing recordings that are too short (too few breathing cycles) according to this measure's definition (there should be at least 640).
To be consistent with the docs, I think it would suffice to change line 234 from ```if len(bbi) > 65:``` to ```if len(bbi) > 640:``` (or to ```if len(bbi) / 10 > 64:```, similar to how it is written for DFA_alpha1 in line 225, for a reason I don't really understand).

I was hesitant to suggest a concrete change since I'm not really familiar with these measures and I am just raising this point based on what I noticed. Apologies for that, and please let me know if I'm still not clear.

closed 2025-02-25 by stale[bot]