neuropsychology/NeuroKit #1040: 1kHz default for ecg_peaks

francu-pixel, 2024-10-19:
Hi, I would like to ask you an explanation on why the ecg_peaks function correctly works for sampling frequency values of 1kHz, while it seems to introduce more errors in the peak detection for different sampling frequencies. I read the code related to the function and I can't understand why this happens. If you have a theoretic explanation of this behaviour I would be very grateful.

DerAndereJohannes, 2024-10-19:
Hi!
The sampling frequency always depends on the input of your signal. If your signal is 1000Hz, then you have to use `1000` as the sampling_rate in *every* function. If it is 250Hz, then you would use `250` in the function input.

Here is an example program where you can see the differences between sampling rate. Here you can see the peaks are all looking pretty good. Of course, this is synthetic data.

Therefore, if you are having an issue with a signal you have in particular (or I misunderstood your issue), please give more information including sample code and some of the data if possible.

```python
import neurokit2 as nk
import matplotlib.pyplot as plt

sampling_rates = [100, 250, 500, 1000, 2000]
signals = []
peaks = []

# Generate signals and compute peaks
for sr in sampling_rates:
    raw = nk.ecg_simulate(duration=20, sampling_rate=sr)
    cleaned = nk.ecg_clean(raw, sampling_rate=sr)
    _, peaks_info = nk.ecg_peaks(cleaned, sampling_rate=sr)

    signals.append(cleaned)
    peaks.append(peaks_info)

# Plot graphs
fig, axs = plt.subplots(len(sampling_rates), 1)

for i, sr in enumerate(sampling_rates):
    axs[i].set_title(f"ECG Clean signal with detected R peaks (sampling_rate={sr})")
    axs[i].plot(signals[i], label="Clean Signal")
    axs[i].scatter(x=peaks[i]["ECG_R_Peaks"], y=[signals[i][j] for j in peaks[i]["ECG_R_Peaks"]], color="red", label="R Peaks")

fig.legend(["Clean Signal", "R Peaks"], loc='upper left')
plt.tight_layout()
plt.show()

```
[image]

francu-pixel, 2024-10-20:
Sorry, I probably didn't correctly explain myself.
This is my experience: I had a signal which was acquired with a sampling frequency of 20kHz and when using the ecg_peaks function, the peaks weren't correctly found. After resampling the signal at 1kHz and applying the function on the new re-sampled signal, the peaks were all correctly found. 
What I wanted to know is if this issue is related to how the ecg_peaks function is written, or if just that too much detail is given and so the peaks are incorrectly assigned as a general issue.

DerAndereJohannes, 2024-10-20:
20 kHz is a very large sampling rate for ECG signals. How noisy is the signal? Most R peak detection algorithms use some sort of gradient thresholding, so higher frequency components can significantly impact the thresholding values. I would guess that downsampling your signal to 1000Hz probably got rid of most of your noise (higher frequency components) which allowed the detector to capture all of the correct peaks.

You can actually take a look at the detection thresholds and results by doing:

```python
import neurokit2 as nk
import matplotlib.pyplot as plt

SAMPLING_RATE = 20_000

raw = nk.ecg_simulate(duration=10, sampling_rate=SAMPLING_RATE) # use your signal here
clean = nk.ecg_clean(raw, sampling_rate=SAMPLING_RATE)

nk.ecg_findpeaks(clean, sampling_rate=SAMPLING_RATE, show=True)
plt.show()

``` 

Please let me know if this answers your question or if something doesn't seem right.

francu-pixel, 2024-10-20:
Yes, thank you, this was the confirmation I needed to be sure.

closed 2024-10-20 by francu-pixel