neuropsychology/NeuroKit #1058: IndexError in `_ecg_findpeaks_manikandan()`

cgastaud, 2024-12-16:
**Describe the bug**
In the function, if the window goes outside the signal length, we try (badly) to clip the window, causing an `IndexError`
See [here](https://github.com/neuropsychology/NeuroKit/blob/master/neurokit2/ecg/ecg_findpeaks.py#L1032)

**To Reproduce**
Example steps to reproduce the behaviour. For instance:

```
signal = np.arange(5)
search_window_half = 10
i = 0
highs = np.arange(i + 1, i + search_window_half + 1)
highs = np.delete(
    highs, np.arange(np.where(highs == len(signal))[0], len(highs) + 1)  # there shouldn't be a +1 here
)
```
We could remove the `+1`, but why not simply change `highs` to:
```
highs = np.arange(i + 1, min(i + search_window_half + 1, len(signal))
```
and remove the if block ?

**Expected behaviour**
Not throw an error

DerAndereJohannes, 2025-04-25:
Hi Sorry for the late response. I took a quick look at your proposal and think it looks correct so you may be onto something. The additional if statement does look quite overengineered.

In fact, your suggestion made me realise that even the lows are not 100% correct. It does not do a check for if it goes into the negatives which would mean that it starts taking the points from the end of the signal.

Does this look reasonable to you (with your own changes):

```python3
lows = np.arange(max(0, i - search_window_half), i)
highs = np.arange(i + 1, min(i + search_window_half + 1, len(signal)))
```

If you think so, it would be interesting to test these changes and get them merged. Would you be willing to create a pull request?

cgastaud, 2025-04-25:
we can, but after tinkering a bit with it seems like the whole for loop is not used ? we build the `id_maxes` array but then return `idx` which was computed much earlier, so I'm not sure what exactly we're trying to do here

DerAndereJohannes, 2025-04-25:
Good eye. It seems like this is hinting towards a larger problem. It looks like this method should be re-verified based on the original paper. Let me know if this sounds interesting to you or else I will put it on my todo.

cgastaud, 2025-04-25:
Ok I circled back to the paper and the function of this loop seems to be finding the "True R-peak" (see section 2.4), aka the max amplitude in the vicinity of the detected peaks.
I can try to work on this next week but I can't make any promises

DerAndereJohannes, 2025-04-26:
Thanks for taking the time to look at it. I took a look at the paper and the code in the repository and it seems to be that the entire idx_max loop is redundant and the remaining code does in fact follow the algorithm correctly.

The [line](https://github.com/neuropsychology/NeuroKit/blob/master/neurokit2/ecg/ecg_findpeaks.py#L1023) (right before the loop):
```python3
# Find points crossing zero upwards (negative to positive)
idx = np.argwhere(np.diff(np.sign(zn)) > 0).flatten().tolist()
```

does exactly what is meant in the paper that can be seen in Figure 5 (g) Output of positive [Zero Crossing Point] ZCP finder. You could argue that the accuracy is +- 1 sample as we are working with differences. You could account for this by interpolating between two samples to find the closer output, but this is not done in the paper and seems perhaps too nitpicky. I am at least relieved that the extra code didnt make the answer wrong.

I think the solution is just to remove the redundant code below the above mentioned line. Let me know what you think and feel free to make a pull request for it as it was your find. ;-)

cgastaud, 2025-04-30:
I think you're right about the window size, and I'm ok with leaving it as is. Maybe we could add it as a parameter ?
Finding the local maximum in a window is done a bunch of times in the other peak finding methods, maybe we could reuse one of them ? [Here](https://github.com/neuropsychology/NeuroKit/blob/master/neurokit2/ecg/ecg_findpeaks.py#L806) for instance.

The paper does talk about the "largest amplitude" though, which means local minimums too (I guess in order to account for inverted QRS ?). That makes it trickier. Maybe we could use an approach similar too the prominence delineation method [here](https://github.com/neuropsychology/NeuroKit/blob/dev/neurokit2/ecg/ecg_delineate.py#L805) ?