neuropsychology/NeuroKit #1061: nk.download_zip throws `AttributeError` when invoked

ignatious-p, 2025-01-22:
**Describe the bug**

I ran the following code on Google Colab

```python
import pandas as pd
import numpy as np
import wfdb
import os
import neurokit2 as nk

url = "https://physionet.org/static/published-projects/mitdb/mit-bih-arrhythmia-database-1.0.0.zip"
download_successful = nk.download_zip(url, ".")
```

this returns 

```py
---------------------------------------------------------------------------

AttributeError                            Traceback (most recent call last)

[<ipython-input-33-bf0018a1f393>](https://localhost:8080/#) in <cell line: 0>()
      1 url = "https://physionet.org/static/published-projects/mitdb/mit-bih-arrhythmia-database-1.0.0.zip"
----> 2 download_successful = nk.download_zip(url, ".")

1 frames

[/usr/local/lib/python3.11/dist-packages/neurokit2/data/database.py](https://localhost:8080/#) in download_zip(url, destination_path, unzip)
     64 
     65     # Download the ZIP file
---> 66     download_successful = download_from_url(url, zip_filepath)
     67     if download_successful:
     68         if unzip:

[/usr/local/lib/python3.11/dist-packages/neurokit2/data/database.py](https://localhost:8080/#) in download_from_url(url, destination_path)
     27     response = urllib.request.urlopen(url)
     28 
---> 29     if response.status == 200:
     30         with destination_path.open("wb") as file:
     31             file.write(response.content)

AttributeError: 'HTTPResponse' object has no attribute 'status_code'
```

**To Reproduce**
Run the above code on Google Colab or any Python environment

**Expected behaviour**
Download the files into the directory

**Screenshots**
[image]

**System Specifications**
- OS: Linux (ELF 64bit) 
- Python: 3.11.11 
- NeuroKit2: 0.2.10 

- NumPy: 1.26.4 
- Pandas: 2.2.3 
- SciPy: 1.13.1 
- sklearn: 1.6.0 
- matplotlib: 3.10.0

ignatious-p, 2025-01-22:
I did some debugging and fixed the error by making the following changes to `neurokit2/data/database.py`.

```python

--- SNIP ---

    # if response.status_code == 200:
    if response.status == 200:
        with destination_path.open("wb") as file:
            # file.write(response.content)
            file.write(response.read())
        return True
    else:
        return False
```

I have commented out the earlier code and replaced it with what I did to fix the issue. 

I believe the earlier error messages are due to changes to the [HTTPResponse](https://docs.python.org/3/library/http.client.html#http.client.HTTPResponse) class in Python's HTTP library. 

I have verified that the proposed changes exhibit the expected behaviour of calling the aforementioned method.

If you'd like to, I can create a PR for this.

DerAndereJohannes, 2025-01-22:
It looks like the issue occurred when removing the `requests` dependency in favour for the builtin `urllib` solution [here](https://github.com/neuropsychology/NeuroKit/commit/79b4888c37b988b80477cd0eed37a8123883ed3a). 

Good catch! Please do make a PR.

commit 872c89e8 (2025-02-11): [fix] update data/database.py to use urllib methods. fixes #1061

Refer https://github.com/neuropsychology/NeuroKit/issues/1061 for details
commit 957f681a (2025-02-11): Merge pull request #1063 from ignatious-p/dev

[Fix] update data/database.py to use urllib methods. fixes #1061
closed 2025-04-25 by ignatious-p