neuropsychology/NeuroKit #1066: Fuzzy Option in RCMSE

typn-git, 2025-01-31:
Hi, thank you for your great work on NeuroKit2!

I am trying to calculate RCMSE using NeuroKit2. To examine the effect of the fuzzy option, I applied both complexity_rcmse() and complexity_fuzzyrcmse() to the same EEG data, but both returned almost identical results.

After checking the source code of entropy_multiscale, I noticed that in CMSE, the _phi() function receives the fuzzy flag via **kwargs, but in RCMSE, it does not seem to be passed to _phi().

# entropy_multiscale.py
    else:
        # CMSE
        if refined is False:
            return _validmean(
                [
                    algorithm(
                        coarse[i],
                        delay=1,
                        dimension=dimension,
                        tolerance=tolerance,
                        **kwargs,
                    )[0]
                    for i in range(len(coarse))
                ]
            )
        # RCMSE
        else:
            phis = np.array(
                [
                    _phi(
                        coarse[i],
                        delay=1,
                        dimension=dimension,
                        tolerance=tolerance,
                        approximate=False,
                    )[0]
                    for i in range(len(coarse))
                ]
            )
            # Average all phi of the same dimension, then divide, then log
            return _phi_divide([_validmean(phis[:, 0]), _validmean(phis[:, 1])])`

How is the fuzzy flag handled in RCMSE?
Thanks in advance!

DominiqueMakowski, 2025-01-31:
Mh could you please run some tests to see if it indeed the results between fuzzy and non-fuzzy are the same (which if I understand you suggest might be happening?) we can help see if indeed there's something wrong

typn-git, 2025-02-04:
@DominiqueMakowski Thank you for your response.

I compared CMSE and RCMSE with and without the fuzzy option using simulated data. In CMSE, the fuzzy option significantly changed the values, but in RCMSE, the values remained the same.

```python
import matplotlib.pyplot as plt
import neurokit2 as nk
```

```python
signal1 = nk.signal_simulate(duration=2, frequency=[5, 12, 40])
```

```python
cmsen1_noFuzzy, cmsen1_noFuzzy_info = nk.entropy_multiscale(signal1, method="CMSEn", show=False, fuzzy=False)
cmsen1_Fuzzy, cmsen1_Fuzzy_info = nk.entropy_multiscale(signal1, method="CMSEn", show=False, fuzzy=True)
```

```python
plt.plot(cmsen1_noFuzzy_info["Value"], label="CMSE noFuzzy")
plt.plot(cmsen1_Fuzzy_info["Value"], label="CMSE Fuzzy")

plt.legend()

plt.xlabel("Scale")
plt.ylabel("Value")

plt.title("Comparison of noFuzzy and Fuzzy Values of CMSE")

plt.savefig("CMSE_comparison.png")
plt.show()
```

[image]

    
```python
rcmsen1_noFuzzy, rcmsen1_noFuzzy_info = nk.entropy_multiscale(signal1, method="RCMSEn", show=False, fuzzy=False)
rcmsen1_Fuzzy, rcmsen1_Fuzzy_info = nk.entropy_multiscale(signal1, method="RCMSEn", show=False, fuzzy=True)
```

```python
plt.plot(rcmsen1_noFuzzy_info["Value"], label="RCMSE noFuzzy")
plt.plot(rcmsen1_Fuzzy_info["Value"], label="RCMSE Fuzzy")

plt.legend()

plt.xlabel("Scale")
plt.ylabel("Value")

plt.title("Comparison of noFuzzy and Fuzzy Values of RCMSE")

plt.savefig("RCMSE_comparison.png")
plt.show()
```

[image]

As an experiment, I modified entropy_multiscale.py to pass **kwargs as an argument when calling the _phi() function in RCMSE. 
```python
        # RCMSE
        else:
            phis = np.array(
                [
                    _phi(
                        coarse[i],
                        delay=1,
                        dimension=dimension,
                        tolerance=tolerance,
                        approximate=False,
                        **kwargs,  #Add
                    )[0]
                    for i in range(len(coarse))
                ]
            )
            # Average all phi of the same dimension, then divide, then log
            return _phi_divide([_validmean(phis[:, 0]), _validmean(phis[:, 1])])
```
After making this change and performing the same calculation, the values in RCMSE changed by fuzzy option.
 

[image]

closed 2025-05-06 by stale[bot]