neuropsychology/NeuroKit #1067: Issue when computing rsp_rate from troughs as np.array or list

me-pic, 2025-01-31:
**Describe the bug**
A description of what the bug is. Try coping the last **steps** that caused the error. Don't forget to also copy if possible the **error log** (the message related to the error).

In the description of the rsp_rate function, it is mentioned that troughs can either be list, np.array, pd.Series, pd.DataFrame. However, troughs are extracted in the code as followed:

[`troughs["RSP_Troughs"]`](https://github.com/neuropsychology/NeuroKit/blob/45c9ad90d863ebf4e9d043b975a10d9f8fdeb06b/neurokit2/rsp/rsp_rate.py#L74)

Hence, an error is thrown if a list or an array is specified when calling the function.

**To Reproduce**
Example steps to reproduce the behaviour. For instance:

```
import neurokit2 as nk

rsp = nk.rsp_simulate(duration=60, sampling_rate=1000)
peak_signal, info = nk.rsp_peaks(rsp, sampling_rate=1000)
rate = nk.rsp_rate(rsp, info['RSP_Troughs'], sampling_rate=1000)

---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
Cell In[36], line 3
      1 rsp = nk.rsp_simulate(duration=60, sampling_rate=1000)
      2 peak_signal, info = nk.rsp_peaks(rsp, sampling_rate=1000)
----> 3 rate = nk.rsp_rate(rsp, info['RSP_Troughs'], sampling_rate=1000)

File ~/.virtualenvs/biosignal2/lib/python3.10/site-packages/neurokit2/rsp/rsp_rate.py:74, in rsp_rate(rsp_cleaned, troughs, sampling_rate, window, hop_size, method, peak_method, interpolation_method)
     71     if troughs is None:
     72         _, troughs = rsp_peaks(rsp_cleaned, sampling_rate=sampling_rate, method=peak_method)
     73     rate = signal_rate(
---> 74         troughs["RSP_Troughs"],
     75         sampling_rate=sampling_rate,
     76         desired_length=len(rsp_cleaned),
     77         interpolation_method=interpolation_method,
     78     )
     80 elif method.lower() in ["cross-correlation", "xcorr"]:
     81     rate = _rsp_rate_xcorr(
     82         rsp_cleaned,
     83         sampling_rate=sampling_rate,
   (...)
     86         interpolation_method=interpolation_method,
     87     )

IndexError: only integers, slices (`:`), ellipsis (`...`), numpy.newaxis (`None`) and integer or boolean arrays are valid indices
```

**System Specifications**

It's important that you give us some information about the system you are using. For that you can run:

```python
import neurokit2 as nk
nk.version()
```

```
- OS: Linux (ELF 64bit) 
- Python: 3.10.12 
- NeuroKit2: 0.2.10 

- NumPy: 2.2.2 
- Pandas: 2.2.3 
- SciPy: 1.15.0 
- sklearn: 1.6.1 
- matplotlib: 3.10.0
```

DerAndereJohannes, 2025-01-31:
Good catch! Thank you for reporting this. It seems as if this is a documentation error, though perhaps we should change the code to better reflect the actual documentation and to therefore allow for your use of the function to work. Right now it seems to assume a DataFrame and directly uses the dictionary key. Here is the quick fix for now:

```python
# Remove dictionary lookup
# rate = nk.rsp_rate(rsp, info['RSP_Troughs'], sampling_rate=1000)
rate = nk.rsp_rate(rsp, info, sampling_rate=1000)
```

me-pic, 2025-01-31:
@DerAndereJohannes Thank you for your quick feedback ! Should I open a PR ? Also if your proposed change is implemented, should the doc be updated to remove pd.DataFrame as a valid format ? Just making sure that there is no confusion, and that it will not be possible in that case to directly pass, for example, the output of `rsp_process` function in `rsp_rate` ?

DerAndereJohannes, 2025-02-10:
Sorry for the late response! A documentation change should definitely happen since it is currently incorrect. We can still make the function more lenient and accept all of the below:
- list
- np.ndarray
- pd.Series
- pd.DataFrame (this would still keep the current behaviour and we can just direct it to `df["RSP_Troughs"]`)

This would make the current documentation correct again. If you feel up to the task, feel free to give it a go too ;-) Otherwise, I can do it on your behalf too.

me-pic, 2025-02-10:
Thank you ! I can open the PR :)

DominiqueMakowski, 2025-02-10:
Go for it Marie-Eve and thanks  🙏

PR #1070 (2025-02-11): [Fix] rsp_rate function 

# Description

closes #1067 

# Proposed Changes

I changed the `rsp_rate()` function so that troughs can be one of those type: list, np.array, dict, pd.Series, pd.DataFrame, and added a `test_rsp_rate()` to `tests_rsp.py`

# Checklist

Here are some things to check before creating the PR. If you encounter any issues, do let us know :)

- [x] I have read the [CONTRIBUTING](https://github.com/neuropsychology/NeuroKit/blob/master/.github/CONTRIBUTING.rst#structure-and-code) file.
- [x] My PR is targeted at the **dev branch** (and not towards the master branch).
- [x] I  [... cut]
closed 2025-05-06 by stale[bot]