neuropsychology/NeuroKit #1073: Is There a Tool in NeuroKit2 for Arrhythmia Classification?

sreeshmaraj, 2025-02-12:
Hello NeuroKit community,

I am currently working with raw ECG data and using NeuroKit2 for signal processing. I wanted to know if NeuroKit2 has any built-in functions or methods for arrhythmia classification.

Additionally, if NeuroKit2 does not support this directly, could anyone suggest any Python packages or tools that can classify arrhythmias from raw ECG signals? I would appreciate any guidance or recommendations on this topic!

Thank you in advance! 😊

Cobrachainsaw, 2025-02-13:
Hi @sreeshmaraj,

Unfortunately, NeuroKit2 does not have built-in methods to classify arrhythmia directly from raw ECG signals. However, it provides several functions that can assist in feature extraction, which is crucial for arrhythmia classification.

Arrhythmia classification relies heavily on feature engineering, where you extract temporal, spatial, and morphological features from the ECG signal before training a model.

For example, ecg_process() is a useful function that preprocesses the ECG signal and extracts key features such as ECG_R_Peaks, ECG_Quality, and ECG_P_Onset. While not all of these features are relevant for arrhythmia classification (primarily ECG_R_Peaks), this function gives you a solid starting point. You can further extract features such as mean RR interval, standard deviation, and interval variability to build a robust temporal feature set.

Additionally, signal_decompose() allows you to perform wavelet decomposition (e.g., using emd or ssa) to extract statistical features like kurtosis, entropy, and skewness, which can enhance your classification model.

If you're looking for a library that can classify arrhythmia directly from raw ECG signals, you might want to check out torch_ecg. It provides pre-trained CNN/RNN models specifically designed for ECG-based arrhythmia detection.

I hope this helps! 😊 If you have any further questions or need more recommendations, feel free to ask.

sreeshmaraj, 2025-02-16:
Thank you so much! Could you please provide a tutorial or guidance for the torch_ecg pipeline?

DerAndereJohannes, 2025-04-06:
Thank you very much again for the issue! Unfortunately I do not have experience with the touch_ecg pipeline. I hope you have been able to find a tutorial that has helped you further. 

I will close this issue for now as it no longer has to do with the repository. However, feel free to open new issues if you find some other NeuroKit related problems.

Thanks!

closed 2025-04-06 by DerAndereJohannes