neuropsychology/NeuroKit #1076: Flexibility with nk.events_plot

ShekharNarayanan, 2025-02-19:
**Describe the solution you'd like**
Hi, relatively new here so please pardon my ignorance! 

1. I was looking to plot some EDA activity using the nk.events_plot function- which works exactly as expected thanks to the awesome tutorials you guys have. One thing that I noticed is that there isn't an option to change the layout of the plot. Ideally, I would like the figure size to be adjusted if I wanted to. So even if I tried `plt.figure(figsize=(a,b))` manually the size of the plot remains the same.

2. On the other hand, features like `plt.xlabel()` or `plt.ylabel()` seem to work, and I get those labels on the plot made by the `event_plot` function.

**How could we do it?**
I would perhaps consider an additional input to the function that can override the axes on the plot. Additionally, perhaps a dict input specifying exactly the layout the user wants could be incorporated in the figure that is created?

My code is below:
```
# Preprocess raw eda signal
signal_eda, info_eda = nk.eda_process(eda_data,sampling_rate=2000)

# Extract clean EDA and SCR features
cleaned = signal_eda["EDA_Clean"]
features = [info_eda["SCR_Onsets"], info_eda["SCR_Peaks"], info_eda["SCR_Recovery"]]

# Visualize SCR features in cleaned EDA signal
plt.figure(figsize=(14,8)) # has no effect on the plot
nk.events_plot(features, cleaned, color=['red', 'blue', 'orange'])
plt.xlabel("time") # works and actually shows on the x axis
plt.ylabel("EDA activity") # shows on yaxis
```

[image]

DerAndereJohannes, 2025-02-19:
Hi ShekharNarayanan! Thank you for reporting the issue that you are having. 

The current version of `nk.events_plot` uses the figure that is created based on the Pandas `.plot()` method, so the figure is created in the function as you have correctly pointed out. I double checked the source code and realised that the documentation states that the function should return the figuren, which it seems to currently not do. It is with this object you can adjust any figure and axes settings. I will help fix that for the next version. 

For the time being you can always get access to the current active figure by using `plt.gcf()` and then adjusting your figure and axes settings based on that. If I were to adjust your code, it would be like this:

```python
# Preprocess raw eda signal
signal_eda, info_eda = nk.eda_process(eda_data,sampling_rate=2000)

# Extract clean EDA and SCR features
cleaned = signal_eda["EDA_Clean"]
features = [info_eda["SCR_Onsets"], info_eda["SCR_Peaks"], info_eda["SCR_Recovery"]]

# Visualize SCR features in cleaned EDA signal
nk.events_plot(features, cleaned, color=['red', 'blue', 'orange'])
fig = plt.gcf()
fig.set_size_inches(14, 8)
plt.xlabel("time") # works and actually shows on the x axis
plt.ylabel("EDA activity") # shows on yaxis
```

I currently feel like adding an ax or a dict as a parameter into plotting functions might make it a bit too complex. By using the fig object you change the figure properties in a much more readable manner (also with the code editor) than putting it into arguments. I am happy to hear thoughts about otherwise though.

Let me know of any additional thoughts and thanks again for bringing this to my attention!

ShekharNarayanan, 2025-02-19:
I understand and thanks so much for the quick response!

closed 2025-02-19 by ShekharNarayanan