neuropsychology/NeuroKit #1087: Why does the ecg derived breathing rate computes to negative value?

paudel54, 2025-04-17:
**I had followed neurokit2 docs where there was an example comparing respiration rate between real RSP data and ECG derived data**
However, the chart clearly showed the rate was going negative. Can anyone explain this issue 

[image]

"import neurokit2 as nk

# Get heart rate
data = nk.data("bio_eventrelated_100hz")

rpeaks, info = nk.ecg_peaks(data["ECG"], sampling_rate=100)

ecg_rate = nk.signal_rate(rpeaks, sampling_rate=100, desired_length=len(rpeaks))

# Get ECG Derived Respiration (EDR) and add to the data
data["EDR"] = nk.ecg_rsp(ecg_rate, sampling_rate=100)

# Visualize result
nk.signal_plot([data["RSP"], data["EDR"]], standardize = True)" code

DerAndereJohannes, 2025-04-20:
Hi paudel54! Thanks for submitting your issue.

The `ecg_rsp()` function tries to extract the Respiration data from the ECG signal (ECG-Derived Respiration or EDR). So the function you see is not actually showing any rate, but the signal which it tries to compare to the recorded RSP.

I can see that the issue may have been the documentation which mentions that the function returns `array – A Numpy array containing the derived respiratory rate.`. I will take care of fixing this documentation issue.

I hope this helps. Let me know if you need anything else.

DerAndereJohannes, 2025-04-23:
Yep! You should be able to use any of the functions that you would normally use on any RSP data (like in your case [rsp_rate](https://neuropsychology.github.io/NeuroKit/functions/rsp.html#rsp-rate)).

Let me know if you encounter any further issue.

closed 2025-04-25 by DerAndereJohannes