neuropsychology/NeuroKit #1096: Issue on page /_modules/neurokit2/signal/signal_timefrequency.html

youneskhandouch, 2025-05-20:
Problem
The np.fft.ifft() function produces complex output, even if the original signal is real. When this complex-valued signal is passed to scipy.signal.hilbert(), it results in unintended behavior or downstream errors if the imaginary part is not explicitly handled.

# Inverse FFT
    signal_ifft = np.fft.ifft(signal_fft)
    signal_ifft[N:] = 0

    # Make analytic signal
    signal = scipy.signal.hilbert(signal_detrend(signal_ifft))

Suggested Fix
Extract the real part of the IFFT output before applying the Hilbert transform:

signal = scipy.signal.hilbert(signal_detrend(np.real(signal_ifft)))

DerAndereJohannes, 2025-05-20:
Hi youneskhandouch,

Thank you for the find and I agree with you and your solution. I would just say to make it more readable, we can simply already apply `np.real` right after the ifft instead of in the method call:

```python3
signal_ifft = np.real(np.fft.ifft(signal_fft))
signal_ifft[N:] = 0

# Make analytic signal
signal = scipy.signal.hilbert(signal_detrend(signal_ifft))
```

To me this looks more readable and makes it explicit that the signal should be real.

If you like, feel free to create a pull request with the changes or let me know if I should do it.

Thanks!

youneskhandouch, 2025-05-21:
Hi Der Andere Johannes,

Thanks a lot for your feedback, I completely agree that applying np.real right after the ifft to improves readability and clarity.

Feel free to go ahead and make the changes on your end. Also, it would be great to check all the functions used in the code to ensure everything remains consistent with the new update.

Let me know if you need anything from my side.

Best regards,
Younes

DerAndereJohannes, 2025-05-21:
Thank you youneskhandouch.

I will keep this issue open until I have updated and checked any additional functions in the repository.

Any Issue brought up is already a lot of help to the project. If you see anything else, let us know.

Best,
Johannes

closed 2025-05-24 by DerAndereJohannes

youneskhandouch, 2025-05-24:
You're welcome! Thanks for pushing the changes to the dev branch. looking forward to seeing it in the next release! If there's anything else you need, feel free to let me know.