neuropsychology/NeuroKit #1108: Dead link in ressources

Aioshiro, 2025-06-25:
Hi,

just a quick issue to point out that there is a dead link in the website, file [docs](https://github.com/neuropsychology/NeuroKit/tree/master/docs)/[resources](https://github.com/neuropsychology/NeuroKit/tree/master/docs/resources)/[resources.rst](https://github.com/neuropsychology/NeuroKit/tree/master/docs/resources/resources.rst).

"`Neuroimaging tutorials and resources <https://.github.io/tutorials-and-resources/>`_" points nowhere.

Aioshiro, 2025-06-26:
Sure, I'll do it when I've some time, I just don't know what the first link is supposed to be.

DominiqueMakowski, 2025-06-26:
Mmh yeah i don't remember maybe we can just remove it 🤷

DerAndereJohannes, 2025-08-09:
Thank you for bringing this up! :-)

closed 2025-08-09 by DerAndereJohannes