neuropsychology/NeuroKit #1125: UnboundLocalError for `epsilon_onset` in `_onset_offset_delineator` when using `method="cwt"` (ecg_delineate.py)

laiamp, 2025-07-15:
## Bug description
When calling `nk.ecg_delineate(..., method="cwt")`, I sometimes encounter the following error:

UnboundLocalError: cannot access local variable 'epsilon_onset' where it is not associated with a value

This happens intermittently depending on the input signal.

---

## Root Cause
After inspecting the source code in `ecg_delineate.py`, the issue is inside the `_onset_offset_delineator()` function. Specifically, the `epsilon_onset` variable is only defined conditionally, and under some input scenarios (e.g., flat or negative peak heights), it remains undefined.

Problematic code snippet:

```python
if peak_type == "rpeaks":
    if wt_peaks_data["peak_heights"][-1] > 0:
        epsilon_onset = 0.05 * wt_peaks_data["peak_heights"][-1]
elif peak_type == "ppeaks":
    epsilon_onset = 0.50 * wt_peaks_data["peak_heights"][-1]
elif peak_type == "tpeaks":
    epsilon_onset = 0.25 * wt_peaks_data["peak_heights"][-1]

# Later usage regardless of definition
if peak_type == "rpeaks":
    candidate_onsets = np.where(cwtmatr[2, nfirst - 100 : nfirst] < epsilon_onset)[0] + nfirst - 100

DominiqueMakowski, 2025-08-13:
> @gemini-code-assist investigate

Well that was useless

Could you make a PR to add a fix (I reckon defining epsilon_onset for an `else` catch-all condition?)

DerAndereJohannes, 2025-08-17:
I am wondering if the default value of 0 even makes sense here? In the original paper by Martínez et al (2004) here are the following values that they used:

[image]

It seems like the conditions if the peak data is less than 0 is not being handled in the neurokit code at the moment (unless I am missing something?). This could also be the fix to the issue.

closed 2026-02-04 by stale[bot]